Next-Flip/Momentum-Firmware #393: Bug in Frequency Analyzer app

uchu61340, 2025-04-05:
### Describe the bug.

Hello,
I've noticed that the frequency analyzer app hasn't been working properly for some time now. When I try to detect the frequency of objects like my car keys or doorbell, it rarely detects anything, even when adjusting the RSSI.
The problem also occurs with an external card, or with the internal module; I find it much less sensitive than before.
Sincerely,

### Reproduction

The fault is still present in frequency analyzer mode in read raw mode I can see the recording of my signal from my car keys or other but nothing is detected in analyzer mode.

### Logs

```Text

```

daveden3353, 2025-04-05:
Would you fucking niggers stop emailing me?! Seriously

uchu61340, 2025-04-05:
I just formatted my sd card and reinstalled the Xtreme firmware and the frequency analyzer application in subghz works very well and detects all frequencies with external module ebyte e07-433m20s and the internal module. So this confirms the bug of this application in the very new version even with unleashed

Willy-JL, 2025-04-05:
You should never ever install xtreme again, it is more than a year out of date and has severe incompatibilities with more recent firmware that you have installed, for example you don't see the battery charge percent anymore. You will gain nothing from installing such old software, except making problems worse. Downgrading is NEVER the answer, in any context, even outside of flipper.

This is not a bug. If you read the changelog, you would know so. External modules do not have the necessary hardware that frequency analyzer needs to give accurate and consistent results, and thus frequency analyzer doesn't use external modules since a few versions ago. This has always been the case on official firmware, and custom firmwares had enabled it to use external modules without knowledge of these issues. Frequency analyzer is not made to listen to random signals. You are meant to hold your remote up to the top left part of the flipper, within a few centimeters, and press the button, to see the frequency. For that, frequency analyzer works great and as intended. Your previous use of frequency analyzer was incorrect, unintended, and giving inaccurate results, possibly even hallucinations.

closed 2025-04-05 by Willy-JL

uchu61340, 2025-04-05:
Sorry but I don't necessarily have time to read all the changelogs. I have a lot of work on card development. It's really weird because I've never seen a reading error under Xtreme with the external modules E07-433M20S in 100mw and 1w that I make in the frequency analyzer, the frequencies were exactly the same as without the modules! Do you have more details on the so-called bug of the application with an external module? My measuring devices in my lab also tell me the same thing as the old version of the frequency analysis application!!!!

Willy-JL, 2025-04-05:
i never said its a bug. i said that the hardware is not capable of running frequency analyzer correctly. those are 2 very different things.

as the changelog for [mntm-008](https://github.com/Next-Flip/Momentum-Firmware/releases/tag/mntm-008) states:

### Updated:
- Sub-GHz:
  - UL: Frequency analyzer fixes and improvements (by @xMasterX):
    - Enforce int module (like in OFW) usage due to lack of required hardware on external boards (PathIsolate (+rf switch for multiple paths)) and incorrect usage and/or understanding the purpose of frequency analyzer app by users, it should be used only to get frequency of the remote placed around 1-10cm around flipper's left corner
    - Fix possible GSM mobile towers signal interference by limiting upper frequency to 920mhz max
    - Fix duplicated frequency lists and use user config for nearest frequency selector too

and i did not say that the results were wrong, i said
> Your previous use of frequency analyzer was incorrect, unintended, and giving inaccurate results, possibly even hallucinations.

keyword being "possibly even hallucinations". the lack of pathisolate and rf switch means that while frequency analyzer switches frequency incredibly fast, the values from one frequency blend into the other, giving inaccurate and sometimes entirely wrong results. doesnt mean that it couldnt give 1 or 2 relatively true results. but point is, it is largely inaccurate and often entirely wrong, this is not its intended usecase, and external modules are not capable of running frequency analyzer correctly.

and no, external modules cannot be made to support frequency analyzer correctly with the right hardware, because these hardware changes diminish the effectiveness of the radio module in general, which would defeat the whole point of using an external module for this purpose as it would give the same range and effectiveness of the internal module.