Next-Flip/Momentum-Firmware #394: Lagg

sariyan-0, 2025-04-05:
### Describe the bug.

The lock screen is unacceptably laggy after booting up.

### Reproduction

No clue

### Logs

```Text

```

Willy-JL, 2025-04-05:
"unacceptably" is neither a technical nor a professional term, and not objective either. Chances are, you have unrealistic standards, and it is working correctly. But it's impossible to say without seeing the "issue" you're complaining about. Send a video of this, and we can see how to proceed. If you'd prefer, you can send it on discord in a bug ticket.

sariyan-0, 2025-04-06:
I’d like to provide some additional context regarding the issue. When the Flipper device either restarts or boots up normally, it prompts for a PIN, as expected, since I have security enabled. However, during the PIN entry process, the screen exhibits noticeable lag. This performance issue appears to have started following the most recent update of the Xtreme firmware

956MB, 2025-04-06:
Are you seeing "lag" after button presses and before menus open, like pressing `RIGHT` and the passport takes longer than expected to open for example, or are you seeing like a bit of a "fade out" of previous screens when clicking around? Because like Willy also kinda said, "lag" is very general.

sariyan-0, 2025-04-06:
The lag happens consistently after button presses on the lock screen, right after the device boots. The image freezes for a few seconds while I’m entering the passcode between clicks. A friend with the same firmware is experiencing the same issue

Willy-JL, 2025-04-06:
> This performance issue appears to have started following the most recent update of the Xtreme firmware

xtreme has been dead for well over a year. no, it has not happened with any update of xtreme, because there have been no xtreme updates. and this is not xtreme, this is momentum.

> I’d like to provide some additional context regarding the issue

why do you need to use ai through basic conversation? are you unable to form full sentences by yourself?

anyway, as i said a video would be more useful than any explanation.

closed 2025-04-06 by sariyan-0