Next-Flip/Momentum-Firmware #397: USB RPC causing furi_check failure

asin9749, 2025-04-15:
### Describe the bug.

Dev build e7fd9a43 was the last build that did not have this issue. It has first appeared in dev build 4a26ecfb and has continued in each build since then.

When trying to connect via USB RPC the flipper resets itself and displays the following error

[image]

[image]

Note BLE RPC is still working correctly as is USB charging. Restoring to dev build e7fd9a43 or previous the issue disappears.

Please let me know if more detail  or logs are required and I will provide once I return from work trip.

### Reproduction

1. Install any momentum dev build that is of version 4a26ecfb or newer
2. After successful install, attempt to connect to the flipper via USB RPC (tested on momentum site, qFlipper and lab.flipper.net)
3. Flipper will reset and show 
[image]

qFlipper will show
[image]

### Logs

```Text

```

Willy-JL, 2025-04-15:
thanks for the details, exact devbuild commit hashes are very useful, especially when it marks one to the very next one. just to be sure before diving deeper, can you confirm this happens on latest bcff3024 too?

also, what are your settings for lock on boot and allow locked usb rpc?

asin9749, 2025-04-15:
Yes the issue is persistent in bcff3024 as well, I checked this before raising the bug.

- I noticed a new setting "Popup Overlay" under Momentum > Interface > General and toggled that but still same issue.
- When toggling `allow_locked_rpc_usb` to `true` and having the PC attempt RPC connection I end up in a `furi_check failed` and flipper reset loop.
- Toggling `lock_on_boot` also does not change the behaviour

I have attempted all the perms / combs of the above three settings and all still encounter the issue.

My original settings file details below 

```
asset_pack: Mine
anim_speed: 100
cycle_anims: 900
unlock_anims: true
menu_style: 1
bad_pins_format: true
allow_locked_rpc_usb: false
allow_locked_rpc_ble: true
lock_on_boot: true
lockscreen_poweroff: false
lockscreen_time: true
lockscreen_seconds: false
lockscreen_date: true
lockscreen_statusbar: true
lockscreen_prompt: false
lockscreen_transparent: false
battery_icon: 6
status_icons: true
bar_borders: true
bar_background: false
sort_dirs_first: true
show_hidden_files: true
show_internal_tab: true
browser_path_mode: 2
favorite_timeout: 7
scroll_marquee: false
dark_mode: false
rgb_backlight: false
butthurt_timer: 21600
midnight_format_00: true
popup_overlay: true
spi_cc1101_handle: 0
spi_nrf24_handle: 0
uart_esp_channel: 0
uart_nmea_channel: 0
file_naming_prefix_after: false
spoof_color: 0
rpc_color_fg: 0
rpc_color_bg: 16777215
```

Willy-JL, 2025-04-16:
@asin9749 please try with build b7cc3c6 (it should be built in about 10 minutes from now). ofw made a bugfix for cli vcp starting/stopping synchronously, im hoping this is what caused your issue. if it is not that, im not sure how to proceed as neither i nor 956mb could reproduce it while testing with those settings.

Willy-JL, 2025-04-18:
build 2583b19c should have some further bugfixes, please try with that

asin9749, 2025-04-20:
I have tried  `b7cc3c62`, `2583b19c` and `be8be359` all still have this same issue for me.

As you have stated that you and the team are not able to replicate the issue I thought to see what would happen if I started fresh and did the below.

1. Did a firmware repair (as per https://docs.flipper.net/basics/firmware-update/firmware-recovery)
2. Installed a new SD card
3. Installed the latest release of OFW 
4. Redo my customisation in `/ext/subghz/keeloq_mfcodes_user`
5. Test my garage and front door sub-ghz files are working.
6. Test that USB RPC still works (no issues in OFW latest release)
7. Install latest Momentum dev build(s)
8. Test the USB RPC is working (it did not work)
9. Install latest dev build of Unleashed (extra apps) build 1130, commit [19351655910a0e9695e16bcfcb7643e30cecb355](https://github.com/DarkFlippers/unleashed-firmware/commit/19351655910a0e9695e16bcfcb7643e30cecb355)
10. Test the USB RPC is working (no issues with latest unleashed dev build)
11. Install last working Momentum dev build e7fd9a43 - so back to working again

As you are unable to replicate, can you please point me in the right direction towards where I should look at the USB RPC code and the recent changes? Then I will attempt a deep look myself next weekend once I am back home with the rest of my kit, I cant do too much more testing with just the work laptop unfortunately.

Willy-JL, 2025-04-20:
@asin9749 just as a test, can you try (on build be8be359) to take out the sd card and then plug into qflipper?

asin9749, 2025-04-21:
@Willy-JL you are onto something, that worked fine.

 [image]

Willy-JL, 2025-04-21:
what if you plug the sd after flipper is already connected to qflipper?

asin9749, 2025-04-21:
That also is working fine

[image]

Willy-JL, 2025-04-21:
Do you happen to have a flipper WiFi development board?

asin9749, 2025-04-21:
Yes, I will be back home tomorrow and will have access to my wifi dev board then

asin9749, 2025-04-23:
I now have the dev board with me

Willy-JL, 2025-04-23:
A stacktrace / backtrace is best shot of figuring this out.

Install ufbt on the pc, connect the devboard to flipper gpio, connect flipper usb to pc, and also plug devboard usb to pc, then:
- `ufbt update --index-url https://up.momentum-fw.dev/firmware --channel dev`
- `ufbt flash_usb_full` (if this fails due to the same furi_check issue we are diagnosing, install the same devbuild as ufbt is configured with via other means like mobile app)
- Apps > GPIO > ESP > ESP Flasher > Quick Flash > Flipper WiFi Devboard > Blackmagic
- Back out to Desktop
- `ufbt blackmagic`
- When gdb prompt opens, type `c` to continue execution
- Open qFlipper, flipper should freeze and debugger should show the crash
- Type `bt` to get the crash backtrace, send it here

if along the way, flipper seems to be turned off and non responsive, *do not panic*: what is happening is when debugger is connected, flipper will pause execution at boot, to fix the non-responsiveness you need to connect the debugger again and exit it cleanly with `exit` command, which resumes execution and lets flipper know debugger is not attached anymore. sometimes after booting like this, flipper will be on desktop and unresponsive, or missing the bluetooth icon, for this just reboot once manually and it will work fine again (if main cpu core takes too long to start due to execution being paused in debugger, the second bluetooth core will stop waiting and not initialize correctly, so if debugger pauses at boot you need to let it continue execution quickly)

asin9749, 2025-05-15:
From WillyJL

 something you can check quicker to confirm my theory: check if apps_data/cli/plugins has any file that does not begin with cli_ or does not end with .fal
another user had a info_cli.fal which was not supposed to exist and caused issues, if you have something similar too, then the issue is already fixed in dev branch

asin9749, 2025-05-15:
[image]

As WIllyJL suspected there many files in there that should not be

After updating to f55a60b1 the issue has been resolved, thanks to @WillyJL and the ret of the team

[image]

closed 2025-05-15 by asin9749

misterabiad, 2025-05-15:
Schreiben sie mir Bitte nicht wurde gehackt Anzeige ist raus

WillyJL, 2025-05-15:
@misterabiad youll have to disable GitHub notifications yourself. We are not writing emails to you. We are working as usual, but you enabled notifications for the work we do.