Next-Flip/Momentum-Firmware #398: Create 2nd encoded sub file with opposite command

Nvidiacerv, 2025-04-15:
### Describe the enhancement you're suggesting.

I would like the subghz app, when a signal is captured and the protocol is recognised, when decoding and
creating a sub file, create a 2nd subfile automatically with the opposite command in case of up/down, 
on/off, open/close etc. for instance when a intertechno signal is decoded into sub file with the off command,
it should create a 2nd file with the on command for the exact same key etc. this feature will prevent the
user to download the sub file to a pc, modify it and upload it again to the flipper zero.

Anonymous3-a, 2025-04-16:
Do you mean that it would make a duplicate signal with a different name? Just name the button something like "Power".
Do you mean that it would make a signal that does an opposite thing from the captured one? Not possible.

Nvidiacerv, 2025-04-16:
> Do you mean that it would make a duplicate signal with a different name? Just name the button something like "Power". Do you mean that it would make a signal that does an opposite thing from the captured one? Not possible.

just when a signal is captured and decoded, not only create the decoded sub file but also a second sub file with the opposite command. It is possible since the protocol is known. Like with the dooya if a down signal is received, not only create the decoded sub file with the received down signal but also a 2nd sub file with the up signal. the value of the opposite signal is knows since the protocol is known. The same for the intertechno protocol when receiving on or off commands. 

my suggestion is to add a new function in subghz_i.c to generate the opposite command after decoding a raw signal and a binary command (on/off, open/close/ or up/down) is recognized.
it also could effect SubGhzReceiver that prcoesses raw signals, 
subghz_scene_add_to_history_callback to provide two returns (orginal and opposite)
subghz_view_receiver this is the ui to show the two files that can be played or saved by the user

the name of the files i suggest that it includes the command so the user can distinquish them from eachother.

I did try to fork the source and make the modifications myself, but i have trouble compiling it.

LS12crew, 2025-04-16:
Think you would have to capture the other signal too, so if you press up and capture it i don't think (could be wrong) the down signal can just be created as its code could be anything so needs capturing too, but  im a noob so might have it completely wrong just what I was led to believe.

Willy-JL, 2025-04-17:
> just when a signal is captured and decoded, not only create the decoded sub file but also a second sub file with the opposite command. It is possible since the protocol is known. Like with the dooya [...]

this is completely wrong. a protocol being known only means that it known what type of signal it is. the manufacturer can put whatever data they want into a signal. there is no universal agreement that 1 is up and 2 is down. what you want is impossible as it varies from every single device and manufacturer.

> All of the key functionality for creating and adding the opposite command is in a single file:
> scenes/subghz_scene_decode_raw.c

that is absolutely the wrong place to attempt to implement it anyway. this scene is for finding known protocol signals inside a raw recorded signal. if this was actually possible to implement, it would have to apply to normal "read" mode too, not just "decode" after "read raw".

i think it is pretty clear that you asked AI about whether this is possible, and to make it for you. AI will not help you, not here and not in any other usecase. what you want is not possible, and AI is making you look like a fool. and the code the AI gave to you will never work, because AI has no clue how to write code, let alone how to write code for such a complicated and undocumented platform like flipper.

closed 2025-04-17 by Willy-JL

Nvidiacerv, 2025-04-17:
> this is completely wrong. a protocol being known only means that it known what type of signal it is. the manufacturer can put whatever data they want into a signal. there is no universal agreement [...]

Allright tnx for trying to help anyway. Just wanna add this.
Whenever i capture a intertechno signal and it is decoded, i duplicate and edit the file in an editor. That file has the frequency, device id etc stored in the key. i modify the last hex values of the key from xx xx xx xx xx xx xx C0 (Off) to D0 (On) to have the opposite command. To find the hex value for the opposite commands, i looked in the intertechno_v3.c file. Its written as clear text for both the 32bit and 36bit frames.
The same for any captured decoded dooya signal, i change the last two position of the key from 11 (down long) to 33 (up long). In case of the dooya protocol i find the values for the opposite commands in the dooya.c file -> subghz_protocol_dooya_const. the same for the down short and down long signals, using the hex values 3C and 1E

I upload the modified files to the flipper zero and am able to send a working signal.
Just trying to find a way to automate this process which i am already using but manually. Anyway again tnx for trying to help.