Next-Flip/Momentum-Firmware #399: Technical Specification for Flipper Zero Mifare Classic 1K Enhancement

killerbe, 2025-04-19:
### Describe the enhancement you're suggesting.

<html>
<body>
<h3 dir="auto" style="box-sizing: border-box; margin-top: 0px !important; margin-bottom: 16px; font-size: 1.25em; font-weight: 600; line-height: 1.25; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">Describe the enhancement you're suggesting.</h3><span style="color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: inline !important; float: none;">test_original.nfc.json</span><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"></p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><a href="https://github.com/user-attachments/files/19822215/hf-mf-E3D143BD-dump.json" style="box-sizing: border-box; background-color: transparent; color: rgb(68, 147, 248); text-decoration: underline; text-underline-offset: 0.2rem;">hf-mf-E3D143BD-dump.json</a></p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><a href="https://github.com/user-attachments/files/19822232/test_E3D143BD.nfc.json" style="box-sizing: border-box; background-color: transparent; color: rgb(68, 147, 248); text-decoration: underline; text-underline-offset: 0.2rem;">test_E3D143BD.nfc.json</a></p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><a href="https://github.com/user-attachments/files/19822211/96B8A921.nfc.json" style="box-sizing: border-box; background-color: transparent; color: rgb(68, 147, 248); text-decoration: underline; text-underline-offset: 0.2rem;">96B8A921.nfc.json</a><br style="box-sizing: border-box;"><a href="https://github.com/user-attachments/files/19822210/2573B480.nfc.json" style="box-sizing: border-box; background-color: transparent; color: rgb(68, 147, 248); text-decoration: underline; text-underline-offset: 0.2rem;">2573B480.nfc.json</a></p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><a href="https://youtu.be/PUBvl1T7aiY" rel="nofollow" style="box-sizing: border-box; background-color: transparent; color: rgb(68, 147, 248); text-decoration: underline; text-underline-offset: 0.2rem;">Video: Original Tag</a></p><span style="color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; display: inline !important; float: none;">&lt;style&gt; &lt;/style&gt;</span><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><b style="box-sizing: border-box; font-weight: 600;"><span lang="EN-US" style="box-sizing: border-box;">1. General Description</span></b></p><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; color: rgb(240, 246, 252); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; or [... cut ...]
Test Case 1: Working with Standard Keys

Action	Data	Expected Result
Scan original finalized tag with MF3 filter	Key A and Key B: FF FF FF FF FF FF	Successful reading and copying of keys with all sector data
Check UID in the obtained dump	 	Correct UID (2573B480) should be read and saved without distortion
Write data to empty MF3 tag	UID: 2573B480	Tag should be correctly written with all sectors and keys
Verify functionality of written tag	 	Copy should successfully open intercom door similar to the original

killerbe, 2025-04-19:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Test
Case 2: Recovery of Unknown Keys<o:p></o:p></span></b></p>

Action | Data | Expected Result
-- | -- | --
Scan   finalized tag with MF3 filter | UID: 96B8A921, unknown keys | Key   recovery process should start with progress indication
Measure key recovery time |   | Keys   should be recovered within a time not exceeding 2 minutes (comparable to   ikeybase express performance)
Check recovery result |   | Correct   keys should be determined (Key A: EF 41 40 37 DC F6, Key B: B6 DA 90 FE 2D   E0)
Read   sector contents using recovered keys |   | All   sectors should be successfully read and saved to dump
Write   data to empty MF3 tag |   | Tag   should be correctly written with all sectors and keys
Verify   functionality of written tag |   | Copy   should successfully open intercom door similar to the original

</body>

</html>

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b>Test Case 3: Sector Reading Stability<o:p></o:p></b></p>

Action | Data | Expected Result
-- | -- | --
Scan   various tag types with MF3 filter | Set of   test tags from different manufacturers | All   sectors of all tag types should be successfully read without errors or   omissions
Display sector reading status |   | Interface   should display status of each sector with indication of problematic areas
Check retry mechanism |   | When   problematic sectors are detected, system should automatically perform retry   attempts with parameter adaptation
Repeat scanning several times |   | Reading   results should be stable and identical between consecutive scans
Check diagnostic log |   | Log   should contain detailed information about reading process with parameters and   status for each sector

</body>

</html>

Willy-JL, 2025-04-19:
Stop fucking using AI. You (probably) have hands, you are able to type a coherent message explaining what you want. I am not reading all that bullshit

Willy-JL, 2025-04-19:
And why are you talking as if you are our boss laying out a project plan? Fuck off

Willy-JL, 2025-04-19:
Come back talking like a decent human in a new issue and maybe we'll try to understand what issues you're having. Use AI another time and you'll be banned from commenting here

closed 2025-04-19 by Willy-JL