Next-Flip/Momentum-Firmware #400: Technical Specification: NFC Tag Information Application for Flipper Zero

killerbe, 2025-04-19:
### Description of the feature you're suggesting.

### Description of the feature you're suggesting.

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

[TMD-5S](https://github.com/user-attachments/assets/58530c2b-cdd7-4102-91b4-5878d1410d58)

[Video: Check NFC](https://youtu.be/dH3hJeDYR00)

[Nfc Tag Detector.zip](https://github.com/user-attachments/files/19822442/Nfc.Tag.Detector.zip)

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

<link rel=themeData
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/Leaf/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=RU style='tab-interval:35.4pt;word-wrap:break-word'>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>NFC Tag
Info Application for Flipper Zero<o:p></o:p></span></b></p>

<p class=MsoNormal><b>Project Overview<o:p></o:p></b></p>

Parameter | Value
-- | --
Project Name | NFC   Tag Info for Flipper Zero
Project Type | Application   for Flipper Zero device
Priority | High
Status | New

<p class=MsoNormal><b><span style='color:white;mso-color-alt:windowtext'>Objective</span><o:p></o:p></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Develop
an application for Flipper Zero that reads and displays detailed technical
information about NFC tags (similar to Proxmark, TMD-5S, or SMKey), including
filter detection.<o:p></o:p></span></b></p>

<p class=MsoNormal><b>Functional Requirements<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l4 level1 lfo1;tab-stops:list 36.0pt'><b>Core
     Functions:<o:p></o:p></b></li>
 <ul style='margin-top:0cm' type=circle>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Scan ISO14443-A NFC tags and
      other supported standards<o:p></o:p></span></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Display: UID, ATQA, SAK, tag
      type, RATS support status, static nonce info, and user hints<o:p></o:p></span></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Detect filter presence and
      type (MF1, MF2, MF3, OTP, ZERO, etc.)<o:p></o:p></span></b></li>
 </ul>
 <li class=MsoNormal style='mso-list:l4 level1 lfo1;tab-stops:list 36.0pt'><b>UI
     Requirements:<o:p></o:p></b></li>
 <ul style='margin-top:0cm' type=circle>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b><span
      lang=EN-US style='mso-ansi-language:EN-US'>Structured output similar to Proxmark
      console<o:p></o:p></span></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b>Clear
      data hierarchy with separators<o:p></o:p></b></li>
  <li class=MsoNormal style='mso-list:l4 level2 lfo1;tab-stops:list 72.0pt'><b>Ability
      to save scan results<o:p></o:p></b></li>
 </ul>
</ul>

<p class=MsoNormal><b>Non-functional Requirements<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Fast scanning (&lt;3 seconds
     per tag)<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b>Proper
     error handling<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b>Low
     power consumption<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l5 level1 lfo2;tab-stops:list 36.0pt'><b>Standard
     Flipper Zero navigation<o:p></o:p></b></li>
</ul>

<p class=MsoNormal><b>Test Cases<o:p></o:p></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Test
Case #1: Scanning MIFARE Classic 1K<o:p></o:p></span></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l2 level1 lfo3;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Action: Launch app, scan MIFARE
     Classic 1K tag<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l2 level1 lfo3;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Data: ISO14443-A MIFARE Classic
     1K tag<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l2 level1 lfo3;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Expected: Structured display of
     UID, ATQA, SAK, tag type, RATS support, static nonce, and tag validity
     status<o:p></o:p></span></b></li>
</ul>

<p class=MsoNormal><b>Test Case #2: Filter Detection<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l0 level1 lfo4;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Action: Launch app, scan tag
     with known filter<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l0 level1 lfo4;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Data: NFC tag with MF1 filter<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l0 level1 lfo4;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Expected: Display &quot;Filter
     detected: MF1&quot; and other tag information<o:p></o:p></span></b></li>
</ul>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Technical
Issues to Resolve<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Based on
previous implementation attempts, these issues need fixing:<o:p></o:p></span></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l3 level1 lfo5;tab-stops:list 36.0pt'><b>FWT
     Timeout errors<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l3 level1 lfo5;tab-stops:list 36.0pt'><b>ATS
     request failures<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l3 level1 lfo5;tab-stops:list 36.0pt'><b>ATS
     reading errors<o:p></o:p></b></li>
</ul>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>Previous
implementation attempted to determine filter type but encountered errors in
debug logs:<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>122528
[D][Nfc] FWT Timeout<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>122579
[E][Iso14443_4aPoller] ATS request failed<o:p></o:p></span></b></p>

<p class=MsoNormal><b><span lang=EN-US style='mso-ansi-language:EN-US'>122582
[D][Iso14443_4aPoller] Failed to read ATS<o:p></o:p></span></b></p>

<p class=MsoNormal><b>Technical Recommendations<o:p></o:p></b></p>

<ul style='margin-top:0cm' type=disc>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Use NFC libraries from Flipper
     Zero SDK<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Work with low-level API for
     precise scan control<o:p></o:p></span></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b>Implement
     timeouts and retry logic<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b>Add
     detailed logging<o:p></o:p></b></li>
 <li class=MsoNormal style='mso-list:l1 level1 lfo6;tab-stops:list 36.0pt'><b><span
     lang=EN-US style='mso-ansi-language:EN-US'>Study existing implementation
     to understand current limitations<o:p></o:p></span></b></li>
</ul>

<p class=MsoNormal><span lang=EN-US style='mso-ansi-language:EN-US'><o:p>&nbsp;</o:p></span></p>

</body>

</html>

Willy-JL, 2025-04-19:
Why did you let your AI off the leash and onto the open internet?

Willy-JL, 2025-04-19:
Flipper already has an NFC app that does exactly what your stupid AI described, and even more. Go away

closed 2025-04-19 by Willy-JL