Next-Flip/Momentum-Firmware #401: Flipper Zero MIFARE Classic 1K Improvement

killerbe, 2025-04-19:
### Describe the enhancement you're suggesting.

## 1. Introduction

I have tested the Flipper Zero with ISO14443-A MIFARE Classic 1K standard and discovered several serious issues that need to be addressed. These problems are particularly acute when working with tags that have an MF3 filter.

## 2. Issues Discovered

### 2.1 Problems with Standard Keys

I attempted to scan an original tag with an MF3 filter that uses standard keys FF FF FF FF FF FF. Unfortunately, Flipper Zero failed to correctly copy these keys. For comparison, I also tried Proxmark3 easy, but it also performed unsatisfactorily.

What's particularly noticeable is the substitution of the tag's UID. The correct UID of the original tag is 2573B480, but during copying it changes to E68487F3. I tried writing both versions to empty MF3 tags, but neither could open my intercom door.

Interestingly, other devices (TMD-5S and SMKey) with the ikeybase express software successfully handled this task - they correctly identified the keys and created a working copy.

### 2.2 Problems with Unknown Keys

I also tested another tag with UID 96B8A921 (with MF3 filter), whose keys were unknown to me. The key recovery methods built into Flipper Zero proved ineffective - it takes too much time. I tried using various known vulnerabilities (nested, hardnested, darkside), but none of them helped.

However, TMD-5S and SMKey with ikeybase express software very quickly recovered the keys through their server algorithm:

- Key A: EF 41 40 37 DC F6
- Key B: B6 DA 90 FE 2D E0

When I took these keys and used Flipper Zero to write to an empty MF3 tag, the result was successful - the copy opened the intercom door.

### 2.3 Problems with Reading Sectors

I also noticed that Flipper Zero doesn't always reliably read all sectors. This depends on the type of tag - some are read completely, others only partially.

## 3. Required Improvements

### 3.1 Working with Standard Keys

I believe that Flipper Zero should correctly copy tags with standard keys FF FF FF FF FF FF. It's especially important to fix the UID substitution problem - the device should preserve the original UID (in my case 2573B480 instead of E68487F3).

### 3.2 Recovery of Unknown Keys

I propose implementing a fast key recovery algorithm similar to that used in ikeybase express. Existing methods (nested, hardnested, darkside) should be optimized to work more effectively with MF3 tags.

It would be useful to add the ability to send tag data to an external server for quick key recovery (similar to ikeybase express).

### 3.3 Reading Stability

I recommend improving the sector reading algorithm, especially for problematic tag types. It would be worth implementing automatic retry attempts with different signal parameters if the first attempt fails.

## 4. How I Will Verify Improvements

### Test 1: Copying Standard Keys

Original tag with MF3 filter and standard keys FF FF FF FF FF FF, I will scan it and verify that:

- Keys are correctly copied
- UID is preserved correctly (2573B480)
- When written to an empty MF3 tag, the copy works the same as the original (opens the intercom door)

### Test 2: Recovery of Unknown Keys

Tag with unknown keys (for example, with UID 96B8A921) and I will verify that:

- Keys are recovered within a reasonable time (no more than 2 minutes)
- The recovered keys are correct (for example, Key A: EF 41 40 37 DC F6, Key B: B6 DA 90 FE 2D E0)
- When writing data to an empty MF3 tag, the copy works like the original

### Test 3: Reading Stability

I will check various types of tags with MF3 filter and be convinced that:

- All sectors are successfully read
- If there are problems with any sector, the system automatically retries and fixes
- Reading results are stable during repeated scans of the same tag

killerbe, 2025-04-19:
Sorry please, I really did use AI back then, did I write normally now?)

bettse, 2025-04-20:
> [Video: Original Tag](https://youtu.be/PUBvl1T7aiY)

I watched the video and he has his hand up like he's given up and it's less than 10% of the way through the system dictionary attack.

I get the distinct impression the user is confusing the specific system they happen to have with the generic case of all mifare classic

> MF3 filter

This is not something mentioned in the Mifare Classic datasheet, I suspect you confuse some vendor terminology.

Willy-JL, 2025-04-20:
> Sorry please, I really did use AI back then, did I write normally now?)

this is still AI, as i had promised i will ban you accordingly from posting here in the future, but i have the decency to out you on your bullshit:

> What's particularly noticeable is the substitution of the tag's UID. The correct UID of the original tag is 2573B480, but during copying it changes to E68487F3. I tried writing both versions to empty MF3 tags, but neither could open my intercom door.

chances are you are copying to a non-magic card, which doesnt allow changing UID. what you are requesting is physically impossible. buy magic tags of the correct kind, and the NFC Magic flipper app will write to them correctly with the new UID.

> I also tested another tag with UID 96B8A921 (with MF3 filter), whose keys were unknown to me. The key recovery methods built into Flipper Zero proved ineffective - it takes too much time.

that is your problem, not ours. you lack patience, and respect for other people's time.

> I also noticed that Flipper Zero doesn't always reliably read all sectors. This depends on the type of tag - some are read completely, others only partially.

obviously, if flipper is unable to find/crack all the keys, not all sectors will be read. you can find keys in other ways, not necessarily requiring other devices. this is your problem, not ours.

> I believe that Flipper Zero should correctly copy tags with standard keys FF FF FF FF FF FF.

it does. its the first line of the dictionary attack.

> It's especially important to fix the UID substitution problem - the device should preserve the original UID (in my case 2573B480 instead of E68487F3).

as i explained, this is your wrongdoing not using magic tags.

> I propose implementing a fast key recovery algorithm similar to that used in ikeybase express.

and riddle me this batman, how? the process is already optimized as far as it can go on the flipper zero hardware. it is easy to talk out of your ass about "make it faster". you are not god, and we are not magicians.

> It would be useful to add the ability to send tag data to an external server for quick key recovery (similar to ikeybase express).

and how the fuck would that work? flipper doesnt have internet connectivity, the devices you mention use proprietary software which we cannot implement ourselves, and also we are not a company, we do not get revenue from your lazy ass cracking tags, we have no incentive to create such a service even if we could.

> I recommend improving the sector reading algorithm, especially for problematic tag types. It would be worth implementing automatic retry attempts with different signal parameters if the first attempt fails.

you dont know shit about what youre talking about. "signal parameters" means nothing here. you dont even know what youre asking. and as already mentioned, the issue is you not having the keys for the tags you want to copy. it would work fine if you did.

> I have tested the Flipper Zero with ISO14443-A MIFARE Classic 1K standard and discovered several serious issues that need to be addressed.

you still are not our boss, we dont owe shit to you.

> [Video: Original Tag](https://youtu.be/PUBvl1T7aiY)

as bettse already said, you gave up after just a few seconds. thats a you problem.

<br>

---

<br>

it is obvious you have no clue what youre talking about. i suggest you kiss my ass goodbye, and follow these steps to properly read your goofy cards:

https://flipper.wiki/mifareclassic/

closed 2025-04-20 by Willy-JL

Willy-JL, 2025-04-20:
just as a final note, comparing flipper zero and proxmark3, which are tools for learning and research and support a large variety of tags (and with flipper even stuff that is not rfid), to tools designed explicitly for cloning tags, is disingenuous and toxic.

you dont compare an axe to a shovel when your objective is digging a 20ft deep hole.

since you own these fancy tools, use them, and shut up.