Next-Flip/Momentum-Firmware #402: NFC Tag Information Application for Flipper Zero

killerbe, 2025-04-20:
### Description of the feature you're suggesting.

## Project Overview

I need an application for Flipper Zero that will read detailed information about NFC tags, similar to how Proxmark, TMD-5S, or SMKey devices operate. This should be a standalone application that quickly identifies the tag type before proceeding to read it with the scanner.

## Application Requirements

I want the application to:

- Scan ISO14443-A tags (primarily MIFARE Classic 1K standard)
- Display all critical technical information about the tag:
  - UID (unique identifier)
  - ATQA
  - SAK
  - Tag type (e.g., "MIFARE Classic 1K")
  - RATS support status
  - Static nonce information
  - Hints for further actions

It's especially important that the application can detect the presence and type of filter on the tag (MF1, MF2, MF3, OTP, ZERO, and others).

## Current Issues

I've attempted to write this application myself but encountered errors. In my implementation, the application doesn't detect filters like TDM-5S or Proxmark do, and produces the following errors in the logs:

```
122528 [D][Nfc] FWT Timeout
122579 [E][Iso14443_4aPoller] ATS request failed
122582 [D][Iso14443_4aPoller] Failed to read ATS
```

## Development Considerations

- The application should work quickly (no more than 3 seconds per scan)
- It should handle errors correctly
- It should use the NFC libraries from the Flipper Zero SDK
- It should implement retry attempts to improve reliability

I've attached the source code of my implementation for further analysis and development. I would appreciate help in solving these issues!

killerbe, 2025-04-20:
Yes, I spent a lot of time testing the bug with the reader, and couldn't formulate it properly at first. I apologize again.

Willy-JL, 2025-04-20:
refer to the previous bullshit in #401 #400 #399 for why you are full of shit

closed 2025-04-20 by Willy-JL