Next-Flip/Momentum-Firmware #405: Scrolling text for infra red transmit buttons

vlado2222, 2025-05-01:
### Describe the enhancement you're suggesting.

Hi.
I have one suggestion. Please add scrolling text for infra red transmit buttons. Current limit on number of displayed characters is not enough to describe function of some buttons, especially if the buttons have multiple functions depending of status of the receiving device.

Thanks!

956MB, 2025-05-01:
I see what you mean. Nice idea, should be a simple add.

[image]

956MB, 2025-05-01:
This is what were talking about, correct?

[image]

vlado2222, 2025-05-01:
Yes, although I would rather prefer the text to be scrolling in one direction in a loop. However that's just a matter of personal preference. One more improvement I would like to suggest is making those buttons closer together to squeeze in one more button on the screen (if possible).

Thanks a lot!

956MB, 2025-05-01:
> I would rather prefer the text to be scrolling in one direction in a loop

That is just my setting to `MNTM > Interface > General > Text Scroll` being marquee instead of standard. Both affect the scrolling in the buttons, so up to you which you want.

> I would like to suggest is making those buttons closer together to squeeze in one more button

I've messed with this also, but to maintain an aesthetic spacing between the buttons, up/down arrows and the header it can only be so small. Without removing the arrows or header theres not enough room to make it 7. Also the way I've made the header wrap to be able to fully view it makes it need one more space (making it 5). It'd probably be possible to hide the header when not on page 1 to allow for the extra cell, but idkif that'd be the best way to go, and having it scroll the header all the time wouldn't be either.

| 1 | 2 |
|--------|-------|
| [image] | [image] |

Willy-JL, 2025-05-02:
i feel like number of buttons should stay the same, this way muscle memory stays as it is for those used to it and also doesnt get confusing between firmwares. scrolling text is very welcome however, might even be a good idea to make a pull request for it to ofw

956MB, 2025-05-02:
Good points. Would it be better for the header to scroll on a single line maintaining the spacing and 6 buttons, or should the spacing shrink to accommodate the wrapping header? I like the way the wrap looks personally, being able to see it all at once, but whatever works best.

vlado2222, 2025-05-02:
> That is just my setting to `MNTM > Interface > General > Text Scroll` being marquee instead of standard. Both affect the scrolling in the buttons, so up to you which you want.

Ow. I didn't know that. thanks!

> Edit: Maybe theres just enough room. idk if it looks better though.
> 1 	2
> ![](https://private-user-images.githubusercontent.com/39519792/439756453-53d06681-a06d-4ffb-9a4b-a0f58b5a2515.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDYxNDUxNzcsIm5iZiI6MTc0NjE0NDg3NywicGF0aCI6Ii8zOTUxOTc5Mi80Mzk3NTY0NTMtNTNkMDY2ODEtYTA2ZC00ZmZiLTlhNGItYTBmNThiNWEyNTE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTAyVDAwMTQzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEyN2E2ZTdkNWQ5YmJiYzUwOGZlODAzNTVjMTQ1ZTZlODlhZTNhNWZjNGVkN2ZlM2VkMGFjMmVkOWZlMWY4ZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Zn8dzjFs-sVJ_p1BgolVayGjb6vZDF1msU8B64yx07I) 	![](https://private-user-images.githubusercontent.com/39519792/439756457-b6382cdf-d685-4a59-98b6-bec1fc7bfb15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDYxNDUxNzcsIm5iZiI6MTc0NjE0NDg3NywicGF0aCI6Ii8zOTUxOTc5Mi80Mzk3NTY0NTctYjYzODJjZGYtZDY4NS00YTU5LTk4YjYtYmVjMWZjN2JmYjE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTAyVDAwMTQzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgxZTI2NzA0MDkwMzQ0Y2YwMDYwNjg1MTg5MzVmMWIxZTQzZGJjZjc1MzQ4YTllYWI5ZTFkNDY0NTEyZDNiMGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.T9aXZRs9UOXK5JwjSwpGBLjgZ4P4LD-3EYgYcm-YFTo)

Certainly difficult to judge without seeing it on Flipper screen. So idk.

> might even be a good idea to make a pull request for it to ofw

Yes!

vlado2222, 2025-05-02:
One more idea! For those buttons that have scrolling text open the right side of the button to show two more characters and get rid of the three dots. Once it starts scrolling open the left side as well. Basically make it look like the entire button is wide and it's shifting.
[image] [image]

956MB, 2025-05-02:
That isn't going to change because this is the intended behavior of scrollable text and it happens everywhere. The ellipsis are an indication that the text is longer than it's container and that it scrolls when selected. Also with the wide button idea, it's too much unnecessary work to tell when the text is at the beginning/end and change the button shape accordingly, and this *isn't* the behavior of ui anywhere else on the flipper, or in general really.

So this one probably won't go in.

956MB, 2025-05-03:
Opened flipperdevices/flipperzero-firmware#4210 over in OFW

WillyJL, 2025-07-20:
merged in mntm-011

closed 2025-07-20 by WillyJL