Next-Flip/Momentum-Firmware #406: SubGhz chat not functioning

GameLord2011, 2025-05-01:
### Describe the bug.

The command `subghz chat` is taking any keypress as an exit. I am running my fork, but I did not mess with the subghz command, I added some commands and modified the gui. But whenever I run the `subghz chat` command, it comes out like this:
```Bash
Receiving at frequency 433920000 Hz
Press CTRL+C to stop
Oprat:

```

### Reproduction

Run the `subghz chat` command in the cli and press anything, it will take it as `CTRL+C`

### Target

Subghz

### Logs

```Text

```

WillyJL, 2025-05-08:
i see youve fixed this yourself and submitted a patch to official firmware, nice work! i have included it here and it will be in the next devbuild.

GameLord2011, 2025-05-12:
@​hedger said that it'll be merged by the next release cycle.

WillyJL, 2025-05-21:
@GameLord2011 i couldnt find another way to contact you and id rather not cause more fuss in the PR thread on official firmware; im not trying to stand up for anyone else but just want to help you in future interactions. the way you behaved in that thread was unnecessarily rude: everyone makes mistakes, and when its such a simple and obvious one theres no need to go all "remove the stupid ... that broke ..." and "did you know you caused this in the first place?". when making big changes, small mistakes slip through, and the skilled developers that made them perfectly understand the issue and that they made a mistake, rubbing it in their face doesnt help anyone, simply pointing it out politely would be much better; i for one wouldve banned you from my projects if that attitude was shown towards me. i understand youre young and enthusiastic about what youre learning, so just try to keep this in mind, just like youre a human with feelings making a comment there is a human with feelings on the other side reading it, be understanding and have patience. just opening the PR with something like "fix small logic error in subghz chat" wouldve worked fine. also goes for "Is anyone going to merge this? It's been a week since I opened this.", this is open source, in most cases projects on here are unpaid hobbies, flipper zero official firmware is an exception as they develop it as a company with a product, but they still have their processes and priorities; a week is an unrealistic demand for most open source projects, id suggest not "bumping" until a few months have passed with no news, and even then you can ask so more politely.

GameLord2011, 2025-05-21:
K.

closed 2025-05-21 by GameLord2011