Next-Flip/Momentum-Firmware #407: Adding custom fonts guide in wiki

Kuronons, 2025-05-01:
### Describe the enhancement you're suggesting.

The Asset-Packs part in the wiki is lacking the custom fonts aspect, especially on the file format / making part and an explanation of the Flipper basic fonts and their use.
I think it deserves a lil guide (or maybe at least a redirect to unofficial Flip wiki guide, if what is written there is valid with Momentum)

956MB, 2025-05-01:
100% agree. The asset packs page at the moment is literally just the old readme from [here](https://github.com/Next-Flip/Momentum-Firmware/blob/dev/documentation/file_formats/AssetPacks.md) lol. It was a good starting point moving it out of the buried documentation folder and into the wiki, but I wasn't ready or "qualified" to rewrite it yet.

Willy-JL, 2025-05-01:
I'll remake it, the old one focuses too much on "how it works" instead of "what it is and how to make one" anyway