Next-Flip/Momentum-Firmware #409: Marantec D832 - 433Mhz

KrisJanssen, 2025-05-03:
### Description of the feature you're suggesting.

I have one of the listed transmitters but pricing for additional ones is frankly ludicrous… what would it take to create emulation of this particular style of transmitter? Currently only an 868Mhz model is supported.

956MB, 2025-05-03:
The [official docs](https://docs.flipper.net/sub-ghz) have pages on sub-ghz and how to read/emulate and so on. Or you could look through the [official](https://flipperzero.one/discord) or [momentum](https://discord.gg/momentum) discords, since there may already be some info on this in there if you look around.

closed 2025-05-03 by 956MB

KrisJanssen, 2025-05-03:
I do not agree @956MB : the pages you refer to have generic info on subGhz use cases while it is very clear that on this project, there is a single person developing most of the emulation code, an activity which clearly requires some knowledge. 

I was asking what kind of information or input or information on a given type of transmitter would be needed to put development of a specific emulation in the backlog.

956MB, 2025-05-03:
If the those docs don't have __all__ of what you needed or I misunderstood, then sorry. The second part of my response was the important part.

While this is an *issue* for you, you're not talking about a bug with or a feature to the firmware itself, which is the point of these. While yes Willy is heavy lifting on the project, theres also far more experts on this stuff who know how to help in the places I've just recommended to you, and over there is the best place to discuss and get what you need. It's not on one person to stop the other things they're working on to figure out someone else's specific use case, not here at least.

I wasn't saying shut up and don't ask, I was saying *heres* the best way to figure it out... the "forum", Discord.

KrisJanssen, 2025-05-03:
Ok. No worries, I will have a look at the Discord.

Willy-JL, 2025-05-03:
While I do pull most of the maintenance weight here and develop my fair share of stuff, I have never developed a subghz protocol. That is more the unleashed team's expertise. I vaguely remember hearing marantec recently, did you try on momentum 010 to see if it was added recently? If so and it is not supported, you could either ask unleashed team by providing a raw recording of pressing each button a couple times, or you could look into developing one yourself, which doesn't have much documentation nor tutorials, there are some Derek Jamison videos on the topic but they don't cover much of the reverse engineering process you would need, which is the hard part, and they also don't focus much on rolling codes from what I remember; best shot for developing your own is looking at existing protocols code, and learning how to reverse engineer the data sent on your own. Or just kindly ask unleashed team without expecting anything, just like us they do this out of passion and not as a job.