Next-Flip/Momentum-Firmware #412: Sending SubGHZ using External CC1101 causes crashes

avipars, 2025-05-09:
### Describe the bug.

on v10, and when i try to use my external cc1101 board to send subghz signals... the flipper crashes and says furi check failed... it worked fine in version 9

### Reproduction

1. Switch on
2. Open Sub-Ghz app
3. Click saved
4. Send any file
5. Flipper crashes

### Logs

```Text

```

avipars, 2025-05-09:
Spectrum Analyzer crashes too

WillyJL, 2025-05-09:
cant replicate. you might have a faulty module or a corrupted install. try reinstalling the firmware, if that doesnt solve it its probably a faulty module. if you made the module yourself, check the connections; if you ever connected it directly to 5v without a stepdown to 3v3, you destroyed the chip. if not, then maybe you have your connections wrong, or its just a dead module.

0xbennyv, 2025-06-20:
I can replicate this issue, I had a poorly soldered CC1101 with a bridged 3&4 pin. so as @WillyJL said this is 100% due to a faulty module.

avipars, 2025-06-21:
Is it repairable?

avipars, 2025-06-21:
> cant replicate. you might have a faulty module or a corrupted install. try reinstalling the firmware, if that doesnt solve it its probably a faulty module. if you made the module yourself, check the [...]

I tried a few different firmware issues. Most likely is a hardware problem as @0xbennyv stated

closed 2025-06-21 by avipars