Next-Flip/Momentum-Firmware #414: Suggestion about Donations and Nostr

StellarStoic, 2025-05-11:
### Describe the enhancement you're suggesting.

The world is going in a digital direction, and you might consider changing that legacy Bitcoin address you have at the bottom to Bech32 (starting bith bc1q) because TX are cheaper or even better, add a lightning wallet for microdonations like willy@momentum-fw.dev. for example if someone want to donate you 5000 sats, it's not economically feasable making this transactions onchain but using Lightning network it's faster, way, way, way cheaper and people might donate more, Micro-transactions are a thing nowdays. If any of you reading this is on Nostr, then you know what I'm talking about.

5000 sats might not sound much today, but who knows what the future holds?

Setting up your lightning address is stupidly easy. Create a new lightning wallet in [walletofsatoshi](https://www.walletofsatoshi.com). You can use that given address of the lightning, but if you want to redirect the payments from something like willy@momentum-fw.dev - follow this guide on how to set up the .well-known folder  https://orangepill.dev/bitcoin-guides/guide-to-create-lnaddress-redirection-on-your-domain/ 

Optionally join [Nostr](https://njump.me/), set up your lightning address, publish releases there, and receive zaps (real Bitcoin micropayments) I like to call them appreciation payments through lightning directly into your LN wallet you set up earlier. 

Sounds complicated, but it's really not. 

As I mentioned. Just a suggestion to join V4V (value for value) economy.

Anonymous3-a, 2025-05-12:
As a note: Bitcoin and most other cryptocurrencies are super polluting, and honestly I'd recommend not even having crypto payments. I do understand most people don't like online payments, though (as I don't like them either).
Just a note though.

StellarStoic, 2025-05-12:
> Bitcoin and most other cryptocurrencies are super polluting...

Has been debunked multiple times and what you said clearly is not true. Majority Bitcoin miners are using green energy and it's known Bitcoin miners can be actually beneficial for the grid in certain circumstances when there's surplus of energy. 

Online payments in fiat like paypal, Patreon, kofi all take their cut. What for? if that can be avoided with technology we have why not?

I'm speeking only about Bitcoin and Bitcoin is money. Better money. Which should not be mixed with other useless noise out there.

Online payments are fine but there's a better way of doing things without 3rd parties. We have tools to avoid watching our savings getting decimated by the current monetary system and that's Bitcoin.

In my lifetime I experienced 2 government  issued currency collapses and I know I'm gonna see at least one more, so yes, I'll always advise anyone to save in the hardest asset in the universe.

WillyJL, 2025-05-12:
i only added a bitcoin address as its the most established crypto and im not a cryptobro, for that same reason im not gonna play around with whatever goofy new hype tech youre shilling here. i will consider adding a few alternative established cryptos, like primarily usdt on trc20 chain as from my experience thats has pretty economic gas price even on low transactions

StellarStoic, 2025-05-13:
Touché

closed 2025-05-13 by StellarStoic