Next-Flip/Momentum-Firmware #416: Furi_check Failed error

ACI077, 2025-05-18:
### Describe the bug.

The furi_check failed error appears. I downloaded several applications from the roguemaster firmware to momentum, and the furi_check error started popping up, and the Flipper itself rebooted. How do I add new apps and how do I fix this error?

Thank you in advance

### Reproduction

1. Install apps from RogueMaster to Momentum

### Logs

```Text

```

WillyJL, 2025-05-18:
Why would you expect applications from another firmware, especially one as unstable and crash-prone as roguemaster?

Install apps using https://lab.flipper.net/apps

Other apps on github can be installed by compiling with ufbt

WillyJL, 2025-05-18:
If by then some applications still crash, that's just an issue with the app, apps aren't perfect

closed 2025-05-18 by WillyJL