Next-Flip/Momentum-Firmware #417: NFC emulate ISO15693-3 Flipper crash and reboot

rygerek, 2025-05-19:
### Describe the bug.

During NFC ISO15693-3 emulation of a certain access card, FLIPPER crashed and was rebooted. I think I tried all the possibilities: disabling animations, debug mode and everything that can be changed in the settings. The strange thing is that the emulation of the ISO15693-3 card works fine, without hanging the Flipper only on XTREME software

### Reproduction

Card emulation NFC ISO15693-3

### Logs

```Text
I don't have a LOG file
```

WillyJL, 2025-05-19:
Details...

What card is it? What crash is it? What are you doing exactly that crashes? Share the file in question maybe?

rygerek, 2025-05-20:
Card is NFC ISO15693-3  read earlier FLipper Zero. Flipper crashes and reboots when trying to emulate it. To simulate a terminal reading this card, you can use an android phone and Taginfo program. When trying to read with the phone, the same situation occurs with flipper - it crashes and reboots. Best Regards

[9 Technicy.zip](https://github.com/user-attachments/files/20343823/9.Technicy.zip)

WillyJL, 2025-05-21:
@rygerek i cannot replicate the crash with that file. what version are you on? what is the crash message?

rygerek, 2025-05-21:
@WillyJL  the latest available version of Momentum 10. I also tried restarting Flipper to factory settings and clearing the MicroSD memory. Flipper continues to crash with the message: “Flipper Crashed and was rebooted. furi_check failed.” It crashes during emulation, but only when I apply to the reading device.

peach098, 2025-12-05:
> Update to dev

I've updated to dev. It chashes too but now the error is "furi_check failed"

aaronjamt, 2025-12-10:
> does not crash for me with the provided nfc file [@peach098](https://github.com/peach098)

Interesting that you're not able to reproduce it, it still crashes for me on the latest commit of dev (98a1a3de6d42e0735d1cb3fe837f55e3b54ba815). The UART logs aren't helpful (even at TRACE level):
```
25385 [D][Iso15693Parser] Frame parse failed

[CRASH][NfcWorker] furi_check failed
        r0 : 100
        r1 : 2001f8de
        r2 : 100
        r3 : 102
        r4 : 20021388
        r5 : 2
        r6 : 2001f8de
        r7 : 200213f8
        r8 : 1
        r9 : a5a5a5a5
        r10 : a5a5a5a5
        r11 : 20031364
        lr : 803c249
        stack watermark: 7720
             heap total: 189648
              heap free: 49376
         heap watermark: 48632
        core2: not faulted
System halted. Connect debugger for more info
```

 Here's a screenshot of the call stack (this is from an older commit 63081d3fc4b7d9700d9ddbe902294a5063fe5c45 due to some issues I'm having with my devboard, but it's definitely failing in the same spot on the latest commit):

[image]

It's specifically failing [here](https://github.com/Next-Flip/Momentum-Firmware/blob/dev/lib/toolbox/bit_buffer.c#L325-L327), somehow the `iso13239_crc_append` is trying to add data past the end of the buffer.

aaronjamt, 2026-01-01:
> [@aaronjamt](https://github.com/aaronjamt) a quick check could be increasing `ISO15693_3_LISTENER_BUFFER_SIZE` by 2 bytes or more in `lib/nfc/protocols/iso15693_3/iso15693_3_listener.c`.

Looks like that helps. The `tx_sequence` also needs to be expanded accordingly, otherwise it crashes later. This patch does fix the crash for me:

[iso15693.patch](https://github.com/user-attachments/files/24398724/iso15693.patch)