Next-Flip/Momentum-Firmware #428: Improved "Input buttons vibrations" option

Zakrzewiaczek, 2025-06-24:
Hi, I have a small suggestion regarding the vibration option when interacting with buttons - in my opinion it would be better if the vibration was triggered only when the button was pressed (not when released, similarly in e.g. smartphones).

Zakrzewiaczek, 2025-07-25:
I noticed a small bug in this tab (_Input_), if _Buttons Vibro_ is OFF, then after switching to value 1 the motor will not vibrate (and it should with the lowest power), and when returning directly from this value to OFF, it will vibrate slightly (even though it should not now)

956MB, 2025-07-25:
> I noticed a small bug in this tab (_Input_), if _Buttons Vibro_ is OFF, then after switching to value 1 the motor will not vibrate (and it should with the lowest power), and when returning directly from this value to OFF, it will vibrate slightly (even though it should not now)

Which version are you on, because 0e3e1b35 this is not the case at all for me. Or are you misunderstanding when the vibro occurs? Like going from OFF to 1 only you'd feel nothing because it just got set, and going from 1 to OFF you'd feel it because it WAS on 1.

WillyJL, 2025-07-25:
to confirm, change the option and then validate what i does by pressing up and down in the menu. works as expected here.

Zakrzewiaczek, 2025-07-25:
Sorry, at first I thought that the settings were saved only after exiting