Next-Flip/Momentum-Firmware #443: GuiSrv crash while decompressing bitmap icon

aaronjamt, 2025-07-22:
### Describe the bug.

While running 65ff6249aaff9db45eddf814f54939d251c0b4a3, I've encountered a system crash twice caused by the GUI service attempting to draw the desktop. I have never encountered this before `mntm-11`, although I'm not sure exactly which commit might've caused this. Below is a screenshot of the call stack after crashing:

[image]

My first thought is that it might have something to do with the connected debugger, possibly due to being paused in the middle of the decompression routine and unpaused after a delay?

### Reproduction

I don't have exact reproduction steps, I'm opening this issue mainly to document it. I'll update it with more information if I can reproduce it.

### Target

Flipper Zero

### Logs

```Text
[CRASH][GuiSrv] lib/toolbox/compress.c
        r0 : 0
        r1 : 0
        r2 : 200001a8
        r3 : e5
        r4 : 200087b8
        r5 : 200392ec
        r6 : 0
        r7 : 80
        r8 : 0
        r9 : 2
        r10 : a5a5a5a5
        r11 : 20031364
        lr : 803c66d
        stack watermark: 1292
             heap total: 189672
              heap free: 130024
         heap watermark: 112464
        core2: not faulted
System halted. Connect debugger for more info
```

### Anything else?

This is probably a low priority bug, especially if it's due to a connected debugger.

WillyJL, 2025-07-22:
youre using a broken asset pack, some animation frame is larger than 128x64. or the animation was compiled 3+ years ago when fbt compiled them slightly wrong in some edge cases.

closed 2025-07-22 by WillyJL

aaronjamt, 2025-07-22:
> youre using a broken asset pack, some animation frame is larger than 128x64. or the animation was compiled 3+ years ago when fbt compiled them slightly wrong in some edge cases.

I don't think any of these are the case, but I could be wrong. I installed (with resources) yesterday from a Github Actions build `.tgz`, so it shouldn't have any outdated/broken/too large assets.

Probably does make sense to keep this closed so it's not filling up your tracker until I find a consistent way to repro.

aaronjamt, 2025-07-22:
> what do you have in momentum > interface > graphics > asset pack

I'm using the "Momentum" pack.

WillyJL, 2025-07-22:
can you reproduce it by going to desktop and holding ok a bunch of times to change animation?

aaronjamt, 2025-07-22:
> can you reproduce it by going to desktop and holding ok a bunch of times to change animation?

I held OK (via Bluetooth) like 20 times and it didn't crash, although the debugger is no longer connected and I only have access to it via the mobile app currently. Will try again when I can connect the debugger and use the physical button.