Next-Flip/Momentum-Firmware #445: Duplicate setting

Zakrzewiaczek, 2025-07-25:
### Describe the enhancement you're suggesting.

Both in:
`Momentum Settings > Interface > General > Clock Midnight Format`
and in:
`Settings > System > Time Format`
There is an option to choose a 12-hour or 24-hour clock format, but I think this is unnecessary (it's unlikely that anyone will use one clock in a different format than the other).

956MB, 2025-07-25:
The `Clock Midnight Format` setting is specifically for choosing the format of the 12-hour clock between 12-1AM, as discussed in #317 and #341 . It doesn't apply to the 24-hour clock because 12-1AM is always 00:XX. Someone using the 12hr clock may want either "12:30" or "00:30" for personal or regional preference. So it's not a duplicate setting.

WillyJL, 2025-07-25:
also, for the record, some thing are in fact duplicated in the momentum app in order to make them easily accessible, for example the "show clock" option in statusbar section and in desktop settings. these 2 however as 956MB explained are different things.