Next-Flip/Momentum-Firmware #446: SubGHz "Smart Jam" mode

AeternusSomniafonsConcordiae, 2025-07-25:
### Description of the feature you're suggesting.

### **Problem Statement**
Current jamming modes lack feedback, forcing users to blindly transmit noise. Meanwhile, read-only modes can’t actively disrupt signals. A hybrid approach would enable:
1. **Verified jamming** (visual confirmation of signal disruption)
2. **Targeted attacks** (jamming triggered by specific signal patterns)
3. **Opportunistic capture** (grabbing codes between jams or pre-trigger).

### **Proposed Modes**
#### **1. Burst Jam + Read (Time-Sliced)**
- **Behavior**: Rapidly alternates between:
- *Jamming burst* (5–20ms TX, full power on target freq)
- *Listening window* (1–5ms RX, attempts to capture signal)
- **Use Case**:
- Verify jamming effectiveness via real-time signal corruption on Flipper’s UI.
- Capture *repeating* codes (e.g., doorbells) during listening gaps.
- **Limitations**:
- Captured signals will be fragmented (UI should mark as "noisy").
- Not suitable for long/protocol analysis during jamming.

#### **2. Selective Jam (Trigger-Based)**
- **Behavior**:
1. Passive RX until predefined signal is detected (e.g., specific protocol, RSSI threshold, or saved pattern).
2. Immediately jams target frequency upon detection (blocks 1–2 full transmissions).
3. Returns to passive RX after timeout.
- **Use Case**:
- Precisely block car keyfobs, alarms, or other critical signals without continuous TX.
- Logs triggered events (time, signal snippet, protocol).
- **Advantage**: Minimal power use, avoids blanket interference.

#### **3. Offset Jam (Experimental)**
- **Behavior**:
- RX on target frequency (e.g., 433.92MHz).
- TX noise on ±0.5–2MHz offsets (e.g., 433.90/433.94MHz).
- **Use Case**:
- Partial disruption of wideband receivers (cheap remotes, some sensors).
- **Limitations**:
- Less reliable than direct jamming. Requires higher TX power.

### **Technical Considerations**
- **Hardware Limits**: CC1101 cannot RX/TX simultaneously on same freq. Time-slicing must account for radio settling delays (~1ms).
- **UI Feedback**:
  - Clear state indicators (e.g., "JAMMING [BURST]", "LISTENING", "TRIGGERED JAM ON [PROTOCOL]").
  - Signal graph shows corruption during jamming (e.g., blanking/RSSI spikes).
- **Safety**:
  - Default to *low-duty cycle* in time-sliced mode to avoid overheating.
  - Warning if user enables offset jam (inefficient power use).

### Anything else?

I'm a bit of a noob to RF, so I don't know if this is feasible.

Anonymous3-a, 2025-07-26:
Momentum Firmware does not support jamming. It is illegal and also just not very nice.

WillyJL, 2025-07-26:
> Momentum Firmware does not support jamming. It is illegal and also just not very nice.

Couldn't have said it better myself

closed 2025-07-26 by WillyJL